nefarius/FireShock #1: BSOD after fresh install

c0dycode, 2016-09-12:
Hey nefarius :)

I've been testing FireShock a bit. Now all of a sudden I got a BSOD after a reboot. Aaaand it was caused by FireShock.sys.

This was kind of satisfying I guess :P Since a friend of mine tested it out on Windows 7 and got a BSOD aswell on Saturday. I'm not 100% sure if it's still needed to change the TargetSystem since I've read the WDK 10 seems to be downwards compatible.

To get to the point:
The last driver I was able to get working without BSOD and a fresh install was commit "Implemented DS4 to XUSB translation" 7708497b526712e3e724d7c5945cd096dc341481

Here's a picture of the BSOD-Overview :)

http://i.imgur.com/6bTLYF4.png

If you need any more infos, just go ahead and ask here or on twitter. :+1:

charlesmilette, 2016-09-12:
A copy of the Minidump will greatly help @nefarius to diagnose this issue.

~~So finally I made a good decision by not installing FireShock now~~

c0dycode, 2016-09-12:
Totally forgot to add that I'm on Windows 10 1607 (Anniversary Update) x64

[091216-18468-01.zip](https://github.com/nefarius/FireShock/files/467135/091216-18468-01.zip)

Here's the dmp-file :)

c0dycode, 2016-09-14:
A little update from my side :)

So, the FireShock driver doesn't seem to be the problem. When I remove the ViGEm driver it works. And the BSOD my friend had was probably caused by me compiling it wrong. Can't test that though, since he's only online on the weekend :) Will see if I can find out which commit causes the ViGEm problem :)
Will leave this issue here open for now though :) Just in case!

c0dycode, 2016-09-14:
@nefarius If I revert the last two commits over on the ViGEm Sources it works perfectly fine.

So the issue is introduced with https://github.com/nefarius/ViGEm/commit/984d3f198ce1c8b2dea7f0c07369c74cc6eec52c for me. Since ViGEm on it's own runs without any problem and FireShock is basically what causes it, I posted the update in here :)

nefarius, 2016-09-18:
Hey hey, thanks for testing and reporting. I think the crash is caused by my unfinished driver-to-driver interface; ViGEm currently lacks reference counting so the BSOD is probably caused by ViGEm getting unloaded before FireShock. I'll look into this.

nefarius, 2016-09-18:
Btw. please always make a full kernel memory dump and test with debug builds, that'd help a lot.

c0dycode, 2016-09-19:
Oh good to know. Switched over to Kernel dumps now and will use the debug-builds when I install both next time :)

nefarius, 2016-09-19:
I've just pushed the latest changes; in my tests no more BSOD if you use FireShock in conjunction with ViGEm 😃

c0dycode, 2016-09-19:
Nice :) Will test as soon as I can :)

c0dycode, 2016-09-20:
Okay, got to it just now. And so far both work fine without BSOD on the newest version. So I guess it's fixed :) Thanks

nefarius, 2016-09-24:
Nice :smile:

closed 2016-09-24 by nefarius