nefarius/FireShock #3: Seems to work flawlessly but game become laggy after few minutes

Sephiroth98, 2016-10-08:
Hi, thanks for the driver, just tried with Street Fighter IV Ultra on steam, all fine but after few minutes the game a bit unresponsive, stutter background music, etc. If I unplug and plug the usb cable again everything seems back to normal, but not for long, few more minutes got the same issue again.

nefarius, 2016-10-08:
Are you using it in conjunction with ViGEm? What are your system specs? CPU load during gameplay?

sun7, 2016-10-19:
I met the same issue.
I'm using FireShock with ViGEm.after a while,game gets lag with out voice and my game controller doesn't work.so,I quit the game and found my PC still working. I opened the TaskManger.
[image]
            <<<up here is process Windows Audio Device Graphics Isolated>>>
[image]
            <<<it's my Audio Device currerntly that time >>>
I'm using a Faked (so called= =) dual shock 3 controller without a sound output device.

Hope this will get help 👍

c0dycode, 2016-10-19:
That's something he's already investigating. I told him about the sound device on twitter.

I'm using a real Sixaxis DualShock3. This happens for me, when I press UP on the DPAD. You could try if that's also the case for you.

nefarius, 2016-10-22:
One issue, one topic please. The sound device is caused by ViGEm not by FS.

nefarius, 2016-11-07:
@Sephiroth98 [This should fix it](https://github.com/nefarius/FireShock/files/574048/FireShock_Setup_1.1.0.exe.zip)

nefarius, 2016-11-12:
Any updates on your end? Could you test the linked release?

closed 2017-01-17 by nefarius