nefarius/FireShock #4: Nav Controller

Encryption767, 2016-10-08:
Would it be possible for you to support the PS3 Move nav controller with these drivers? Or maybe make them recognized as a ds4 to be picked up in DS4Windows for the auto profiler it has. 

I have used it with scptoolkit but there is no direct input only or convert to direct input option and the profiler has always been broken. 

Excellent controller to use with FPS games in combination with a mouse like the g600 or naga, especially when your PC is in the living room and not on a desk (eliminates having to keep a keyboard in your lap),, but just can't seem to find any software what would let me map the whole controller out to kb or kb and xinput hybrid for the axis.

nefarius, 2016-10-08:
If I'd have one or at least a machine with one connected then sure I could 😄

mirh, 2016-10-08:
thp/psmoveapi#185
For as much.. I dunno, it'd feel both weird and wrong to have two open source drivers for the same device.

nefarius, 2016-10-08:
Oh, if there already is one I don't wanna interfer 😃

Encryption767, 2016-10-08:
Wasn't able to do anything with thp/psmoveapi#185. Think that is a lib being made for other programs to incorporate?

Anyways, nefarius has software already that lets one connect and work perfectly with xinput with the scptoolkit. Unfortunately, there are games out there that for some silly reason will not allow a controller/kb/mouse combo. Without being able to turn off the virtual 360 driver and just map the controller to keybinds instead then it can't be used. It works out that you have to have either or and aiming with a thumb stick sucks balls.

He has a very simple driver here though. I thought it would be nice to see a nav controller connect and stay connected so maybe it can be mapped with something like xpadder or antimicro.

nefarius, 2016-10-08:
@Encryption767 if you force it over your controller via Device manager does it work?

Encryption767, 2016-10-08:
Scptoolkit or FireShock? Trying to install Fireshock to the navigation controller entry in the device manager ends with with a power failure code.

nefarius, 2016-10-08:
Ah I see 😆

nefarius, 2016-10-08:
Well if I'd have one I'd gladly take a look. I always accept hardware gifts though 😉

Encryption767, 2016-10-08:
USB\VID_054C&PID_042F&REV_0100
USB\VID_054C&PID_042F

Hardware ids. 

You could probably grab one for $5 at gamestop lol.

nefarius, 2016-10-08:
I'm already devoting a **lot** of time and knowledge to my free and open source projects but I fail to see why **I** should also buy the hardware to analyze 😉

Encryption767, 2016-10-08:
It connects perfectly to SCPtoolkit. Great work with that btw. If SCPtoolkit had the ability to just spit raw data or map to directinput or xinput instead of xinput or nothing and had a good profiler then I could drop ds4windows and just use scptk for my ps controllers :P

Encryption767, 2016-10-08:
If you were in the states then I would.

mirh, 2016-10-08:
> Oh, if there already is one I don't wanna interfer

Not that I think you should really bother with this, given we are all lusting over polished fireshock and airbender.. But according to the other ticket, it should totally behave just like a DS3 (but sensors).

nefarius, 2016-10-09:
@Encryption767 I made some modifications from memory, don't know if it works or crashes, you're invited to test 😄 Feedback welcome.

[FireShock_Setup_1.0.2.exe.zip](https://github.com/nefarius/FireShock/files/518297/FireShock_Setup_1.0.2.exe.zip)

Encryption767, 2016-10-09:
Great work. Anti micro picked everything up right away and I was able to map the device properly. Xbox controller also was picked up and was able to be mapped. 

There is an issue with DPC latency however. With the Fireshock driver installed, the latency begins to climb. I am still testing to see when it happens though. I want to rule out antimicro before I can say for sure.

Latency spikes.
[image]

Default.

[image]

Encryption767, 2016-10-09:
Ok. Just tested with the controller plugged in with antimicro running without the fireshock driver installed and all is fine in DPC land.

[image]

Encryption767, 2016-10-09:
Just tell me if there is something in particular you need me to test and I will help however I can.

nefarius, 2016-10-09:
Thanks for the feedback. Since I'm fairly new to kernel land it's entirely possible that I've goofed something up. I'll check out the LatencyMon tool on my systems and report back.

Encryption767, 2016-10-09:
Sounds great. If you can clear that up then Fireshock will certainly be well on its way to being a preferred light weight driver.

nefarius, 2016-10-09:
Does the slowdown appear with or without ViGEm or does it not matter?

Encryption767, 2016-10-09:
It appears with and without.

nefarius, 2016-10-09:
Ok, thanks!

Encryption767, 2016-10-09:
Anytime!

nefarius, 2016-10-09:
Hm, what's that on your end?

[image]

Encryption767, 2016-10-09:
The same for FireShock.sys. USBPORT.sys goes nuts though.

[image]

nefarius, 2016-10-09:
Ha! Well that's something, thanks 😄

Encryption767, 2016-10-09:
Sys specs-
EVGA x79 Dark
EVGA 1080 SC
Corsair Vengeance 32gb ddr3 
Intel Core i7  4820k @ 4.4hgz

Encryption767, 2016-10-09:
USBPORT.sys has tripled since I posted that. I kept Latency mon running since that last post.

[image]

nefarius, 2016-10-09:
WTF have I done 😕

Encryption767, 2016-10-09:
That's just a DPC count. Could be my system tbh.

I still have it installed and for now all is well. But I am watching it. I am going to try to use it in combination with antimicro.

Encryption767, 2016-10-09:
Something is up for sure. I have mapped a regular ps3 controller out with antimicro but then I closed it out and checked latency mon in the background..

[image]

Encryption767, 2016-10-09:
Not sure if this means anything but Kernel-PnP event log shows an error.

Device USB\VID_054C&PID_042F\6&1aad3f20&0&2 could not be migrated.

Last Device Instance Id: USB\VID_045E&PID_028E&IG_01\2&9C6DEB5&3&01
Class Guid: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Location Path: 
Migration Rank: 0xF000FFFF0000F122
Present: false
Status: 0xC0000719

johnrossporter, 2016-10-31:
How did you get Device Manager to recognize the FireShock driver on a PS3 Move Navigator Controller? I tried to edit the  inf with the correct hardware ID's, but it complains about the hashes not being present in the  catalog.

Encryption767, 2016-10-31:
He posted a link to a special version within that thread. Unfortunately, it was doing something wrong under the hood as the DPC latency climbed up and up as long as the driver remained.

nefarius, 2016-10-31:
Once these issues are resolved the next release will support the nav controller.

Encryption767, 2016-10-31:
Sounds great nefarius!

nefarius, 2016-11-06:
Oi, I've fixed the latency issues 😄 

[image]

Nav Controller also integrated; ViGEm interface needs a bit of work then a new release is ready for publishing.

Encryption767, 2016-11-06:
Looking forward to it!

nefarius, 2016-11-06:
You can test this pre-release if you want: [FireShock_Setup_1.1.0.exe.zip](https://github.com/nefarius/FireShock/files/574048/FireShock_Setup_1.1.0.exe.zip)

**Caution:** don't use with current release of ViGEm or system might crash!

Encryption767, 2016-11-06:
I don't have ViGEm installed.

I gave it a test and for the first 5 min all was well.

[image]

Unplugging and plugging the device has settled the latency down now so I am going to use it for a bit and see if it climbs back up.

Encryption767, 2016-11-06:
Played about 5 min of q3 before it became unbearable. Tried with a regular ds3 controller and have the same issues.

mirh, 2016-11-06:
Then.. I'd *assume* this and #3 are duplicate.

Encryption767, 2016-11-07:
On another note.. when I try to build the driver myself in VS2015, It can not find driver.h from device.h.

If it isn't included with your git version then on your next commit could you include driver.h?

mirh, 2016-11-07:
To compile KMDF/UMDF drivers, you need the [Windows Driver Kit](https://developer.microsoft.com/en-us/windows/hardware/windows-driver-kit)

Encryption767, 2016-11-07:
I have both 8.1 and 10

mirh, 2016-11-07:
Perhaps you need to add the correspondent Include folder?

Encryption767, 2016-11-07:
A quick search reveals no driver.h file. I think this is a file he wrote.

mirh, 2016-11-07:
[Doesn't seem like](https://msdn.microsoft.com/it-it/library/hh706187(v=vs.85).aspx). 
Uh.. Maybe we are talking about 7.1 SDK ?

Encryption767, 2016-11-07:
I certainly hope not.

Encryption767, 2016-11-07:
I never said! I am on Windows 10 x64 with the anniversary update.

Encryption767, 2016-11-07:
Which folder do you have to include mirh?

mirh, 2016-11-07:
I'd guess whatever "Include" one contained in the wdk installation folder. 
Inside Program files (x86)\Microsoft SDKs or Program files (x86)\WDK iirc.

JoshWobbles, 2016-11-07:
Sweet, And I was just going to ask if support was coming for the Nav controller!

nefarius, 2016-11-07:
@Encryption767 @mirh sorry for the confusion; you also need to clone ViGEm into the same parent directory for the build to succeed, the include settings are a bit messy at the moment 👼 

@Encryption767 please ensure you test with version 1.1.0, I can not replicate the latency issues anymore on two different machines of mine:

[image]

Encryption767, 2016-11-07:
Check out the amount of time you have it connected.

This is after about 4 min or so. I have the driver version showing here.
[image]

And this is the same session right after uninstalling the driver. I have the device manager open here to show.

[image]

The original one you sent me earlier in this thread was also 1.1.0. Is there anyway you might have gotten them mixed up with each other and sent me the old one instead?

Encryption767, 2016-11-07:
Very strange. What version of windows are you running? Is it x64?

nefarius, 2016-11-07:
My Windows 10 version (x64):

[image]

Try this:

1. Uninstall FireShock
1. Download [RAPR](https://driverstoreexplorer.codeplex.com/)
1. Run as Administrator
1. Select all `Nefarius Software Solutions` entries, check force delete and remove them
1. Re-plug your controller, if it's still showing up as FireShock Device, right click and uninstall from Device Manager
1. Reboot
1. Try again; this is the working build: [FireShock_Setup_1.1.0.exe.zip](https://github.com/nefarius/FireShock/files/576572/FireShock_Setup_1.1.0.exe.zip)

Good luck!

Encryption767, 2016-11-07:
Maybe something is wrong on my end then. I am having this issue still. If someone else would chime in and test then that would be good. Got to play a game with it for at least 4 to 5 min though.

nefarius, 2016-11-07:
Does it occur only with the Nav Controller? Do you have a DS3/4 for testing? I don't have a Nac one so I can't replicate a 100%.

Encryption767, 2016-11-07:
It happens on both the ds3 and the nav. Freed up a USB 3.0 port and I am trying on it now. The previous driver did the same thing on both but so far the usb 3.0 is working properly with the controller. 

Even after 15min, it is still stable.

[97 more replies]

closed 2017-12-25 by nefarius