nefarius/FireShock #6: Not working with Mayflash PS3003

utjduo, 2016-10-30:
Hi! I'm using the Mayflash PS3003 to connect up to 4 PS3 controller wireless. The controller is connected to the PS3003 and the PS3003 spits out a Dinput through USB to the computer, so the computer feels the controller being plugged in through USB.

It feels like FireShock would work flawlessly with the adapter as it gives the computer raw Dinput from the PS3 controller but when I install the FireShock drivers on the Mayflash controller I get:
`This controller cannot start (Code 10)`
and then nothing more.

I don't know exactly what might be wrong. Probably something with that it's not a PS3 controller directly but the PS3003 in the middle, even tho the signals are exactly the same.

The VIDs and PIDs for the controllers are:
```
HID\VID_0079&PID_1801&REV_0100&Col01
HID\VID_0079&PID_1801&REV_0100&Col02
HID\VID_0079&PID_1801&REV_0100&Col03
HID\VID_0079&PID_1801&REV_0100&Col04
```

I hope it can help you to make this work as it would help many many users that uses PS3003 both directly to the computer and trough streaming-devices like the SteamLink together with VirtualHere.
Thanks!

charlesmilette, 2016-10-30:
If your PS3003 spits out DirectInput to your computer, there's no need
for FireShock.

utjduo, 2016-10-30:
But the Dinput gives me messed up controllers and not correct at all. If I change to Xinput it works but then only 1 controller can be connected at ones.

nefarius, 2016-10-30:
I neither know nor support this device and like @charlesmilette already assumed; it doesn't sound like FS is the solution to your problem.

closed 2016-10-30 by nefarius