nefarius/FireShock #7: Disable FireShock -> ViGEm interaction?

JoshWobbles, 2016-11-07:
I saw reference to a possible feature in the Cockpit for disabling this interaction, but seems as of yet not incorporated. The ability to break the direct communication would be helpful.

nefarius, 2016-11-07:
Yes, I do indeed have planned to include some control tools too but they're still in a very early experimental stage.

JoshWobbles, 2016-11-07:
Sweet, first noticed the issue when using this in my new version of IM cause a loop where DS4's would spawn 360 controllers then my app would spawn a DS4 for the 360 and it would loop until BSOD. Took me a minute to realize the 360's were spawning from within FireShock and ViGEm.

nefarius, 2016-11-07:
Well... oops 😄

nefarius, 2017-05-13:
It's included: 
* Key; `HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\FireShock\Parameters`
* Value: `UseViGEm` (`REG_DWORD`)

If the value doesn't exist or is less than 1 direct communication is disabled which is currently the default until I've sorted out all flaws.

closed 2017-05-13 by nefarius