nefarius/FireShock #8: Only one controller connected, but stuck as controller 2?

jacobmix, 2016-11-08:
I got only one controller connected, but it's stuck as controller 2.
A way to change controller number or something would be nice.

Using Windows 10 Home 64-bit.
And here's my devices if that helps:
[image]

JoshWobbles, 2016-11-08:
XInput itself determines "player" numbers, not applications. Unplug all your controllers then go into your devices manager and try removing all instances of HID Game controller, Wireless controller, and 360 controllers in there. Then plug in only your DS3 and with all the phantom or ghost controllers gone it should drop to player one.

nefarius, 2016-11-12:
Due to the XUSB architecture it's not possible to force a controller slot number since the system decides this depending on controller connection order. So you should be fine with @JoshWobbles suggestion.

nefarius, 2017-01-22:
Well, this idled long enough...

closed 2017-01-22 by nefarius