nefarius/FireShock #10: Unable to retrieve MAC address of Dualshock 4 in USB mode

JoshWobbles, 2017-01-20:
The code when using the controller without Fireshock works for getting the serial/Mac:

`                byte[] buffer = new byte[16];
                buffer[0] = 18;
                device.ReadFeatureData(out buffer, 18);
                s = String.Format("{0:X02}:{1:X02}:{2:X02}:{3:X02}:{4:X02}:{5:X02}", buffer[6], buffer[5], buffer[4], buffer[3], buffer[2], buffer[1]);`

but does not work when Fire shock is installed.

JoshWobbles, 2017-01-21:
It appears the issue is with the fact that all Fireshock devices regardless of connection are returning 64 as their `InputReportByteLength` which is what was previously used to determine connection type. Is there a workaround for this?

nefarius, 2017-01-21:
Funny, DS4Windows has no problem with that. I'll look into this.

JoshWobbles, 2017-01-21:
The fallback behavior in DS4W may be a little more lax, my extensions for determining what methods to obtain MAC are a little more strict due to the additional devices I have been adding support for

nefarius, 2017-12-25:
Version 1.x is obsolete and so are related issues. v2 will be published soon™.

closed 2017-12-25 by nefarius