nefarius/FireShock #11: Add support for the new controller that comes with the PS4 Slim

charlesmilette, 2017-02-11:
Right now it won't install automatically and attempting to force install the driver results in a failure with error STATUS_DEVICE_POWER_FAILURE

nefarius, 2017-04-10:
If I'd own one I'd take a look at it 😸

mirh, 2017-04-29:
https://patchwork.kernel.org/patch/9367441/ ?

nefarius, 2017-05-13:
@mirh thanks!

nefarius, 2017-05-28:
I've added it, could someone verify and give feedback please.

charlesmilette, 2017-05-28:
Will do once I get USB passthrough working on my VM

charlesmilette, 2017-06-16:
I never got passthrough working and I don’t want to install a self-compiled driver on my machine because of all the secure boot stuff. I guess I’ll wait for a signed driver release

nefarius, 2017-06-16:
@charlesmilette I'll ship a new signed build then 😉

charlesmilette, 2017-06-20:
Any news on the signed build? ViGEm got one, but not FireShock (which is
the thing I need to test)

JoshWobbles, 2017-06-20:
Seems like he may be holding off because of this: https://github.com/nefarius/FireShock/issues/4#issuecomment-309850075

mirh, 2018-01-30:
https://downloads.vigem.org/fireshock/stable/2.0.0.0/windows/

sylveon, 2018-01-31:
Yeah after a few discussions on Discord I've been using that with Shibari for a while.

closed 2018-01-31 by sylveon