nefarius/FireShock #12: FireShock doesn't completely uninstall itself

Alia5, 2017-02-16:
After installing / uninstalling Fireshock, DS3 Controllers still get shown as "FireShock 3" Device.
This even holds true when installing the "official driver" via the PSNow software.

The controller works as expected though (or doesn't work without any driver, ofc.)

My best guess would be a leftover registry value, but I didn't check any further...

[image]

charlesmilette, 2017-02-16:
The driver didn’t got removed, you can do so with the driver manager by right-clicking on the device.

De : Peter Repukat
Envoyé le :16 février 2017 11:12
À : nefarius/FireShock
Cc : Subscribed
Objet :[nefarius/FireShock] FireShock doesn't completely uninstall itself(#12)

After installing / uninstalling Fireshock, DS3 Controllers still get shown as "FireShock 3" Device.
This even holds true when installing the "official driver" via the PSNow software.
The controller works as expected though (or doesn't work without any driver, ofc.)
My best guess would be a leftover registry value, but I didn't check any further...

Alia5, 2017-02-16:
I tried to do that, didn't change anything.

Upon closer inspection, though. The issue is not with FireShock, but rather ViGEm's HidGuardian-Driver...
[image]

I guess this can be "moved" now...

closed 2017-02-16 by Alia5