nefarius/FireShock #14: Latest Release doesn't work with latest release of ViGEm

RibShark, 2017-04-12:
The latest release of FireShock (v1.0.1.0) doesn't seem to work with the latest ViGEm release (v1.5.8.0). Installing ViGEm v1.0.0.0 causes the FireShock XInput device to appear, while with v1.5.8.0 no such thing happens. I know there have been commits made to FireShock since the latest release on GitHub, but these haven't been compiled.

JoshWobbles, 2017-04-12:
there are also some newer releases in this thread; https://github.com/nefarius/FireShock/issues/4

charlesmilette, 2017-06-04:
Same happens here. I can see ViGEm's virtual Xbox 360 controller, however no input at all is sent on it, while the FireShock controller device works as expected.

nefarius, 2017-06-16:
Yeah call me API-buster, I'll look into this next 😜

sylveon, 2017-09-02:
@nefarius ping

nefarius, 2017-09-02:
Yeah yeah sorry but got no time, currently redesigning parts of ViGEm...

252rer, 2017-10-11:
FireShock 1.3.1.0 with ViGEm v1.13.0.0 Draft version(there are 2 1.13.0.0 releases), causes the device to freeze when plugged in. It shows in normal devices but being blank in game controller settings while also hanging all devcon commands until a reboot. I'm guessing this is due to a mismatch in fireshock's api for ViGEm being to old.

nefarius, 2017-10-26:
This issue is obsolete with the rewrite.

closed 2017-10-26 by nefarius