nefarius/FireShock #15: "Windows cannot verify the digital signature for the drivers required for this device"

Gristrosia, 2017-05-10:
Getting "Windows cannot verify the digital signature for the drivers required for this device" on Windows 10 1703 RS2. I checked the signature on the files, and everything seem to be ok.

Any idea?
[image]

charlesmilette, 2017-05-10:
Turn off secure boot in your bios settings

Gristrosia, 2017-05-10:
Not going to happen.

charlesmilette, 2017-05-11:
Unfortunately due to new signature requirements since the AU the only way to get it working is either have @nefarius get a certificate from VeriSign, having you self-sign it, or disabling secure boot.

Gristrosia, 2017-05-11:
Yes, I read a bit about it but secure boot off isn't going to happen. Thanks anyway.

JoshWobbles, 2017-05-13:
@nefarius Im getting this too on the SCPVB drivers. Cert expired?

charlesmilette, 2017-05-13:
https://blogs.msdn.microsoft.com/windows_hardware_certification/2016/07/26/driver-signing-changes-in-windows-10-version-1607/

we discussed this in nefarius/ScpToolkit#266

JoshWobbles, 2017-05-13:
This is a recent issue though. The cert was always accepted until last couple weeks or so

charlesmilette, 2017-05-13:
Have you done a clean install?

nefarius, 2017-05-13:
I finally know how to fix this, gimme' a few hours 😄

nefarius, 2017-05-13:
Heya @charlesmilette @JoshWobbles @Gristrosia [this should do the trick](https://github.com/nefarius/FireShock/releases/tag/v1.3.1.0).

JoshWobbles, 2017-05-13:
Any chance of a re-signed release of SCP Virtual Bus drivers? Still using that until ViGEm is deemed "stable". 
[image]

nefarius, 2017-05-13:
@JoshWobbles sure, are you using the version from my repo?

JoshWobbles, 2017-05-13:
@nefarius Good question, cant remember, I know I had to do some trial and error before I landed on a stable version as the ones that used libusbk rarely worked and the ones after that when it switched back to the normal USB driver seemed to have OS incompatibilities. I think I may still be on 1.0.2 ish from the original dev branch. When does your fork off at?

nefarius, 2017-05-13:
@JoshWobbles ok nvm then I can build one from scratch and you have a test, sounds more simple, doesn't it.

JoshWobbles, 2017-05-13:
@nefarius I can tell that the driver file version is 1.7.0.0 and is signed with a open source Certum Code cert that expired 3/12/2017.

nefarius, 2017-05-13:
@JoshWobbles yeah then that's toast anyways, [Latest branch should do it.](https://github.com/nefarius/ScpVBus)

Dienfive, 2017-05-15:
@nefarius I followed your instruction on installing fireshock software. but after installing, we cant use it for SCP tool kit, still same issue with Windows driver certificate.
Can u take a look?

Dienfive, 2017-05-16:
hello!

JoshWobbles, 2017-05-16:
@nefarius @Dienfive I created a new issue in ScpToolkit (even though its an idle project) so as not to derail this any further. https://github.com/nefarius/ScpToolkit/issues/759

Dienfive, 2017-05-16:
@JoshWobbles thank you, i go there now

nefarius, 2017-05-28:
Can we close this?

closed 2017-06-11 by nefarius

geter712, 2017-08-20:
Just to add a simple fact. The best and only reliable solution for any driver signature problem is disabling driver signature check within the Windows startup options. I've spent whole eternity trying to find another solution, but everything else is useless. This is not a permanent solution, it works only for current Windows session, which means it needs to be done every time Windows starts. Microsoft purposely made it this way to force driver publishers to pay driver certification. Here's the procedure for all Windows versions http://freewisdoms.com/windows-cannot-verify-the-digital-signature-code-52-driver-fix/

mirh, 2017-08-20:
Actually, it was more about security than anything (at least initially, now with SB requirements start to be nuts)
And it can be permanent without problems with bcdedit nointegritychecks/testsigning