nefarius/FireShock #16: Troubleshooting partial DS3 support

Vebryn, 2017-06-05:
Hi,

I install FireShock to use my DS3 with SteamWorld Heist game. It successfully recognizes my pad. I'm able to use left stick to navigate in the menu but no other buttons are working. Is there a way to see if it's game or driver related ? How to determine if game uses Direct Input, Raw Input or XInput ?

Best regards.

mirh, 2017-06-12:
Well, the first thing to check is windows own controller proprieties window. 

Then, I wouldn't want to be a PITA, but sometimes (?) some internal stuff [dies](https://github.com/nefarius/ScpToolkit/issues/123). 
And nefarius may or may not be finally using an 'improved' design now. 
So.. If I were you and even Windows itself misbehave, I'd try to see in a linux VM with usb passthrough if jstest reports everything working in the first place.

nefarius, 2017-06-16:
Please bear in mind that - until Sony released their "official" sixaxis driver shipped in early 2017 - no game had "native" DS3 support built in. Since I had to customize the report descriptor it's entirely possible that the game can't interpret the inputs it receives. There is no real solution for this other than mapping the pad to XInput and rely on games using this API.

mirh, 2017-06-17:
Isn't "native" DS3 support also using the same [libScePad](https://pcgamingwiki.com/wiki/Controller_talk:DualShock_4#Extendend_native_support_info_5137) library ?
Also, I thought that when having to expose accelerometers and analog buttons, we'd have used that API, if not one similar.

Vebryn, 2017-06-18:
With default driver, Windows doesn't see any button. It neither work ingame.
With ForceShock driver, Windows recognize all buttons, throttles included. It hopelessly cause low latency #4.

nefarius, 2017-12-25:
Version 1.x is obsolete and so are related issues. v2 will be published soon™.

closed 2017-12-25 by nefarius