nefarius/FireShock #19: Controller Working By It Self In Games Only

karim9910, 2017-07-07:
i have a 3rd party dualshock 3 controller.....i used your (ScpToolkit v1.7.277.16103-BETA) and then i used (FireShock v1.3.1.0) the problem is that when i am inside any game like (gta iv,NFS The Run,Fifa 16) the controller will start to work by it self without me pressing any buttons......i thought it must be happening because its a fake controller but when am not playing any game and using the input tester every think works fine and the controller doesnt work by its self anymore but when i enter any type of game i will use the controller fine for the first 30 sec maybe and then it starts working by it self again.please help

JoshWobbles, 2017-07-07:
Those games expect xinput device input format. You need to block the ds3 using HidGuardian then emulate a 360 controller with ViGEm

karim9910, 2017-07-08:
can you please tell me what to do in steps
1_
2_
3_
like that because im not that expert

JoshWobbles, 2017-07-08:
Both of those drivers have readmes that explain installation.

karim9910, 2017-07-08:
[image]
now what

JoshWobbles, 2017-07-08:
Did you read the readme's? That picture tells me nothing.

karim9910, 2017-07-08:
[image]
this is the readme of HidGuardian and the other screen shot is the latest version of ViGEm and i dont know what to do next

JoshWobbles, 2017-07-08:
Read the readme's of ViGEm as well and do what they say.

karim9910, 2017-07-08:
i have discovered something.....the controller will only start pressing buttons by it self when im charging my laptop....and my Touch Pad will start to not work well and the mouse will jump everywhere but i remove the charger and use the laptop on its battery only every thing is fine again,,,controller will work fine and touch pad too.....i think my battery have some issues

JoshWobbles, 2017-07-08:
Hmm, we are getting into the highly theoretical since the only way to prove this may be with an oscilloscope, But the AC/DC adapter may be putting out "dirty" power, or perhaps the battery charge circuit is damaged in the laptop. Both usually dont effect the battery and rarely presents itself unless you have audio equipment hooked up and grounded through the laptop. But if you can confirm that the issue is only when plugged into AC then I think it's a safe bet that the issue likely doesnt reside in the drivers.

karim9910, 2017-07-08:
Im going to remove the battery and use the laptop with ac power only and open any game and tell you the outcome....i think if it works then the problem will be in the battery only

JoshWobbles, 2017-07-08:
Battery is just a battery, and removing that does not narrow the possibility of the charging circuitry since it wouldn't be charging with no battery.

JoshWobbles, 2017-07-08:
Hmmm, Im going to backpedal a tad as I came across a corroborating issue, https://inputmapper.com/support?view=ticket&layout=view&ticketid=145 reported in my bug reporting and has a lot of commonality. @nefarius I maintain that the false inputs are due to the all too common "exclusive mode" issue that games most notably FIFA and Dark Souls have, Im tending to think there may be a common thread somewhere in ViGEm or HidGuardian and PCs in a battery powered state.

JoshWobbles, 2017-07-08:
@karim9910 If you set your power plan while on battery to "Performance", does the issue persist?

nefarius, 2017-12-25:
Version 1.x is obsolete and so are related issues. v2 will be published soon™.

closed 2017-12-25 by nefarius