nefarius/FireShock #20: Ram leak.

Rikimaruaxu, 2017-07-11:
With the latest Fireshock installed, My Ram usage goes up indefinately, With the ps3 controller plugged in it eventually gets to 12-14 gigs of ram used on my 16 gig system, and after unplugging the controller it drops to 476MB, There is a leak somewhere, and It might have something to do with the rumble not working/being super delayed.

mirh, 2017-07-17:
Could you try some of [this](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/using-poolmon-to-find-a-kernel-mode-memory-leak) stuff perhaps?

Rikimaruaxu, 2017-07-17:
Will in a few days. had uninstalled it and returned to using SCP toolkit, gonna finish the ps2 game I'm playing, And then i can use that to see where the leak is coming from in Fireshock.

nefarius, 2017-10-26:
Fixed by rewrite.

closed 2017-10-26 by nefarius