nefarius/FireShock #22: Anatel PS3 controller motion controls and rumble.

subanz, 2017-09-10:
I do believe these are legitimate PS3 controllers, however they were released later 2013-2015.
http://www.psdevwiki.com/ps3/CECHZC2U, listed here under the Overview section.

To make them work without issues in ScpToolkit and ScpServer I had to add this line GenuineMacAddress = 08:A9:5A to the settings folder.

However they also seem to have issues with FireShock and the Sony drivers from the PlayStation Now app.

I think this may because the offsets for data are different for this version of the controller, I've noticed that when using x360ce downloading premade settings assigns the wrong buttons and axis assignments likely because they are different for the older versions that people used to create those settings.

When using ScpToolkit and ScpServer rumble works fine, also installing this mod https://sshnuke.net/cemuhook/padudpserver.html shows all the motion sensors/gyros working properly. However neither work with FireShock. All the other buttons work.

Sorry if I was mistaken and the motion controls don't work on any model of the PS3 controller with FireShock. installing the DirectInput plugin for InputMapper 2.0 did show the motion/gyro axis however none of them had any data which makes me think they are suppose to be working with FireShock but I'm just guessing.

Thanks for any help or info and your hard work on these projects.

nefarius, 2017-10-14:
I'll come back to that once my re-write is done 😄

subanz, 2017-10-14:
Thank you, let me know if you need any additional info.