nefarius/FireShock #26: Is a more Xbox controller 'compatible' HID report possible?

JohnEdwa, 2018-01-01:
The current 'DS4' style one kinda works, but it's far from ideal. Could a custom HID report be made that more closely matched the X360 one?

The usual minor annoyance is that many games will match buttom prompts to what they would be with an Xbox controller, based on the "button x" HID report. For example, X360 maps Button 1-4 to A, B, X, Y, while Fireshock has them as Y, B, A, X, making QTEs pretty impossible. Not to mention L1 and L2 being mapped to Button 11&12, which don't even exist on a X360 controller so they are usually shown as blank.

But I recently hit an even bigger issue with Don't Starve - X360 uses X rot and Y rot as the right stick mapping, while Fireshock uses them for the Left and Right triggers, and they default to 0. This makes it practically impossible to remap the buttons as the moment you enable the controller, it notices X Rot being 0 and starts pressing left. Pressing the triggers half way kinda makes it work, but it's not easy.

I mapped the current ones for the X360 and Fireshock DS3 [on a google sheet](https://docs.google.com/spreadsheets/d/1YlHEekK5u4Pmq-EiNTunGqmY3YkbF1ze8RQyKeVaZjk/edit?usp=sharing) FWIW.

JoshWobbles, 2018-01-01:
Don't think that has anything to do with the HID reports themselves, but more the direct input standard being so abstract from the xinput standard.

JohnEdwa, 2018-01-01:
Maybe I use the wrong terms, but what I mean is this part from the readme:
> If a DualShock 3 gets connected to the USB hub, the filter will inject a modified configuration and HID report descriptor which describes a button and axis layout similar to the DualShock 4. 
The 'button and axis layout similar to the DualShock 4' is not very useful for many games that while not using xinput, kinda assume you use an Xbox controller where "button 1" is A and "button 2" is B, while in Fireshock A is 3 and B is 2 instead.

JoshWobbles, 2018-01-01:
Right, that just makes HID decoding easier for developers, has nothing to do with in game though. Games dont use HID reports.

TheZnayx, 2018-01-01:
The X rot and Y rot corresponding to the Left/Right triggers, which default to 0, causes such things as constantly turning camera in a 3rd-person game. I'm sure people are aware of this, but I just want to reinforce that it looks like JohnEdwa is thinking along the right track. I imagine, though, that if it weren't a difficult issue to fix, it would already be resolved. What is stopping Fireshock from having the inputs mapped more closely to an Xbox360 controller from the PC's perspective?

JoshWobbles, 2018-01-01:
@TheZnayx Again, its the difference in the DirectInput and XInput standard that is causing that. Trying to "fake" the standard would only break the controller for all pure DirectInput games.

TheZnayx, 2018-01-01:
What about adding a DirectInput / XInput toggle? Would that cause issues?

mirh, 2018-01-01:
Also, you should just shortly put on hold complaints, waiting for the (I think) new fireshock version that oughta be released sometimes soon.

JoshWobbles, 2018-01-01:
@TheZnayx essentially thats what using FireShock along with ViGEm does.

JohnEdwa, 2018-01-02:
Isn't the whole point of Fireshock to take the broken original configuration and HID that the controller reports (which [looks like this](https://d1rktuf34l9h2g.cloudfront.net/0/00/00b49358_QZWwPK.jpeg)) and inject its own, fixed one instead?   
Because somewhere in there is the information for "when this bit comes from the controller, it is for button 1, and this other one is for button 2, and this axis is Z rotation" etc, and I just want to fix the order to make them better match the mapping for an X360 controller, so games that use Raw/DirectInput, but assume an X360 controller wouldn't freak out.

mirh, 2018-01-02:
The point of fireshock is fixing the original broken HID descriptor of DS3 
(also making DS4 work with vigem - and possibly allowing to "wire" more of its extensions in the future - but that ain't the point)
You get a \*normal\* controller. End, done. 

A ***360*** controller on the other hand, is \*wait for it\* totally different, and requires whatever it took to implement their stupid never documented XInput api. 
Which falls under the name of Vigem. Check it. 
I see Benjamin in a kind of commits spree today, so again, I'd wait his steam to boil off before nagging him.

JohnEdwa, 2018-01-02:
It's like you see the word "Xbox" or "360"  and just immediately stop reading and assume I want xinput. If the games used xinput then Fireshock wouldn't do anything in the first place.

Look, this is the [default control mapping from 20XX](https://i.imgur.com/wytf3rc.png). 
And [This is how it looks like when remapped with Fireshock,](https://i.imgur.com/8gEc0nQ.png). Note how 1) the face buttons are shown in wrong places, and 2) some of them are just shown as question marks. 
20XX, like *countless* other new games assumes that from the HID report (or whatever it's called), when the controller sends an input for button 1, it's 'A', button 2 is 'B', X Rotation is used for Right Stick Horizontal and so on, following how it is for the x360 pad..

If Fireshock also mapped the "X" to button 1, instead of button 4, and mapped the Right Stick to "X Rotation" and "Y Rotation" instead of using those for the trigger, and so on, it would, first of all, be mostly compatible with default mappings for games, and also not completely messed up with any kind of UI hint or quicktime event.

As far as I know, the current mapping is like it is because it's matching to the one DualShock 4 uses, which is only useful for games that *specifically* have gamepad mapping for a DS4 controller. Which is pretty much three games in the entire universe.

JoshWobbles, 2018-01-02:
@JohnEdwa XInput games read direct input devices too, thats WHY you get any input at all in the game. Would you just trust us already.... Use ViGEm, it is what you are looking for.

JohnEdwa, 2018-01-02:
So there is an official standard that defines for DirectInput that the right analog stick *has to be* mapped to Z Axis and Z Rotation, the triggers *have to be* X and Y Rotation, and the face buttons 1 to 4 *need* to be going clockwise, starting from the top, or it is not valid DirectInput and will break in any DirectInput program?  
Because, for the n-th time, the order of the mapping in Fireshock is the only thing I want to change.

JoshWobbles, 2018-01-02:
@JohnEdwa There is a loose standard that most DI games adhere to, but not all. But the standard differences are more complex than just shifting the order of channels (which again, would break it for games that already support Dualshock Devices, so not actually an option at all), for instance XI shares the trigger axis whereas DI splits them. If you are tired of having to change mappings, then for the love of all things holy just use ViGEm, it is what it is MEANT for.

mirh, 2018-01-02:
> So there is an official standard that defines for DirectInput

Sorry? As in: "in the dx sdk or ddk"?
Because I have docs here at hands if you can point it out. 

> will break in any DirectInput program?

TFW they tell you somebody ever had regards for DI programs breaking.

JohnEdwa, 2018-01-02:
> Sorry? As in: "in the dx sdk or ddk"?
Because I have docs here at hands if you can point it out.

It was a question. As in is there one defining what physical button/axis goes to what virtual mapping, so if the right stick is not Z Axis and Z Rotation and instead is X/Y Rotation it would "brek the controller for all native Direct Input games".
Because AFAIK there isn't, and the only thing it would 'break' is looking like a DS4 controller to games expecting that specific mapping. Just like right now any game assuming buttons are laid out like on an x360 controller they are all messed up, like the 20XX example I gave above.

IDK, maybe even do such a radical thing as make two versions you could choose at installation?

> for the love of all things holy just use ViGEm, it is what it is MEANT for.

But fine. Point me to the 'how to use ViGEm with Fireshock' documentation.  
I installed it following the guide on the wiki, and I see a "Fireshock 3 Device", " HID-compliant game controller" and the "Virtual Gamepad Emulation Bus" in device manager, a "PLAYSTATION(R)3 Controller" in Game Controllers, but it didn't change anything.

JoshWobbles, 2018-01-02:
If you installed both properly, a xinput device should automatically show up when a DS3 is connected per documentation:

> If a DualShock 4 gets connected, it enables XInput-Support if ViGEm is installed. Same goes for a DualShock 3.

Ensure you see the 360 controller in device manager. If so then the next step is to see if the game will let you select that controller (Xbox 360 NOT the Dualshock) and function immediately, or if HidGuardian will also be needed to completely hid the Dualshock from games.

mirh, 2018-01-02:
https://github.com/nefarius/ViGEm/wiki/Driver-Installation

JohnEdwa, 2018-01-02:
> https://github.com/nefarius/ViGEm/wiki/Driver-Installation

That is the wiki guide I followed.

> If you installed both properly, a xinput device should automatically show up when a DS3 is connected per documentation:

ViGEm is properly installed I think, but no 360 pad:

```
PS C:\Users\JohnEdwa> get-installedmodule

Version    Name                                Repository           Description
-------    ----                                ----------           -----------
1.14.0.0   ViGEmManagementModule               nuget.vigem.org      PowerShe...

PS C:\Users\JohnEdwa> Get-ViGEmBusDevice

DevicePath         : \\?\ROOT#SYSTEM#0002#{96E42B22-F5E9-42F8-B043-ED0F932F014F}
InstanceId         : ROOT\SYSTEM\0002
DeviceName         : Virtual Gamepad Emulation Bus
DriverVersion      : 1.14.1.0
Manufacturer       : Benjamin Höglinger-Stelzer
DriverProviderName : Benjamin Höglinger-Stelzer
```
Pluggin the controller in adds the "Fireshock 3 Device" with 'Nefarius Software Solutions 1.3.1.0' driver and a "HID-compliant game controller" that is "on FireShock 3 Device", with default microsoft drivers, but not a 360 pad.

JoshWobbles, 2018-01-02:
Would be in a separate section of device manager, under Microsoft game controllers or something similiar.

JohnEdwa, 2018-01-02:
If it was detected like the actual x360 pad, it would be under "Microsoft Common Controller For Windows Class", but there's nothing.

Devices & Printers and the Game Controllers dialog only shows a 'PLAYSTATION(R)3 Controller' as well :/

JoshWobbles, 2018-01-02:
I vaguely remember Benjamin putting in place a reg key they may enable disable FireShock->ViGEm automatic connecting, anybody remember where that may be?

JohnEdwa, 2018-01-02:
Issue #7.  
I added that and I also tried this on my laptop and with different controller (A PS2 one with a USB adapter), and those didn't work either.

So either there is a step that is required but isn't actually written anywhere (like that registry tweak was), or the Fireshock <-> ViGEm connectivity is broken.  
The other thing I can think of that some remnants of ScpToolbox have survived the CleanWipe utility, uninstallation, and removing the DualShock 3 SCP driver, and they are messing this up.

JoshWobbles, 2018-01-02:
Ahhh, it is possible if you had one of the versions that used lilipad that could bork things up. i'd 86 that. ViGEm forums may be a good resource aswell if there is a reg option or if a recent release did break it. I do know that for a while only certain versions worked with certain versions as both both the public Com's were still being heavily developed and modified.

mirh, 2018-01-02:
*Waiting for new release of just about everything before reporting anything else could also be a good idea*

nefarius, 2018-01-04:
Woha, what is going on here 🤣 

Ok I read the whole thread and have hopefully memorized everything correctly so I can shed some light on this. Here we go:

**TL;DR:** this whole thread is pointless since FireShock v2 is nearly done which will do things completely different. v1 (which still is on GH releases page is **obsolete** and won't be continued anymore due to reasons).

Now let's jump to the past a bit. When I design FS v1 my idea was to modify it's HID Report Descriptor to a more sane developer-friendly format. I was getting away with "inventing" my own descriptor since at that time there was no "official" layout since the DS3 isn't natively supported by Windows and therefore any game. Then Sony silently released their `sixaxis.sys` driver which used a what you might call Sonys "official" layout. Well that was annoying for compatibility but I digress. I then planned and also implemented a driver-to-driver interface which would allow FS v1 devices to directly communicate with `ViGEm` (the magic making an x360 controller appear) so the DS3 would look like an XInput-compatible controller right away so games can pick it up instantly without any special DI/RawInput hacks.

Problem is: driver-to-driver in kernel-mode isn't really a good idea if you want flexibility like changing button/axis mappings and the callback-mechanism for rumble is prone to cause deadlocks and BSOD the kernel. Plus I couldn't ever resolve the DPC latency/memory consumption issues some people had so I basically decided to go for a rewrite. This is upcoming v2 which simply does what a driver is supposed to do: initialize the DS3 via USB so it reports inputs and provide a user-mode API for communication. Nothing more. No self-spawning X-devices, no driver-to-driver BS, just being a device driver.

This change in design also makes big parts of the README invalid and outdated, which I apologize but I simply can't keep up anymore with all the docs and rather spend my time coding and testing.

I hope I covered everything. Piece out.

JohnEdwa, 2018-01-04:
@nefarius Sounds reasonable, I'll go back to ScpToolkit for now and wait for further development.

But the TL;DR of this whole thread is, if at all possible, set up FS v2 in a way that games assuming the x360 layout don't freak out like Don't Starve and 20XX do with FS v1. (linked again, [default mapping](https://i.imgur.com/wytf3rc.png) vs [fireshock mapping.](https://i.imgur.com/8gEc0nQ.png)), because other than that it's been working fine for me. (Installing Fireshock on all my friends' computers when I waltz in with my DS3 instead of the whole ScpToolkit has proven to be a much-preferred solution)

mirh, 2018-01-04:
So....... I'm not sure what would still be the deal of fireshock over the generic sixaxis.sys?

nefarius, 2018-01-04:
FS v2 is an USB driver only, it won't expose HID-Capabilities and won't be visible by games.

nefarius, 2018-01-04:
@mirh I guess you can go for the `sixaxis.sys` route as well if you don't want the Shibari dispatcher.

nefarius, 2018-01-04:
One more note: I have no idea if `sixaxis.sys` supports pairing the DS3 to something else than the PS3, FS v2 does 😏

mirh, 2018-01-04:
> FS v2 is an USB driver only, it won't expose HID-Capabilities and won't be visible by games.

[image]

nefarius, 2018-01-04:
You know, I get complaining from all fronts so I might as well chose the easy route and that is: user-mode driver equals no HID device (and no BSODs).

mirh, 2018-01-04:
_Yeah_, but now I start to feel like 90% of the utility of the project has been thrown out of the window?
Because [afaik](https://github.com/torvalds/linux/blob/e1bc84d0071f59c8b38232e2cb093c47c47e4f9f/drivers/hid/hid-sony.c#L74) ds3 HID is fucked up - and even assuming that could be made to work with windows the same.. It doesn't really seem the basis for a good api.. Or something.

nefarius, 2018-01-04:
I won't expose the original report to anyone :) there will be APIs for that. Or DS4/X360 emulation if you like it simple.

JoshWobbles, 2018-01-04:
@mirh TBH to really get any usefulness out of the Dualshock controllers, I think there will alwayse need to be some sort of application level middleware. Though not being HID compliant is a bummer as that was easy to develop for. 

Also, I gotta ask, what is the draw people have with the DS3? It is one of the biggest pieces of garbage I have ever wrapped my hands around lol. I'd even recommend the Logitech F series controllers over this.

JohnEdwa, 2018-01-04:
@nefarius So is the end goal for Fireshock V2 is to be... 'split' so it'll be FS + Shibari combining to 'ScpToolbox v2', instead of the hacks you are currently doing with v1 to mess with the HID stuff?

@JoshWobbles I've always personally found the PS1 DualShock to be the best for me, and DS3 is just that but with L2/R2 triggers (and lighter/smoother analog sticks, I think). I used a DS2 for a while, but the pressure sensitive buttons are really terrible as analog controls.  
Original Xbox controllers are meh, the x360 has absolutely unusable Dpad and every change Sony made to the DS4 just made it horribly uncomfortable for me. The Xbone controller is really good though, but I find it a little too heavy.
But basically, it's all just 90's kids nostalgia.

mirh, 2018-01-04:
Wat? I could tell you exactly the contrary (especially in the dark times when microsoft was mobbing with their proprietary shit connection). 
Even when motionjoy was the standard.. I'm not sure your logitech things had accelerometers and all?
Also, it's a godsend for ps2 emulation. 

p.s. rewriting HID to have every descriptors where OS expects that x features to be, would certainly have helped a great deal.

JoshWobbles, 2018-01-04:
I get nostalgia, i play SNES roms with a USB SNES controller lol, but nothing ever really took advantage of the selling points for a DS3, (sixaxis/pressure buttons). DS4 feel less "toyish" to me so is usually my go-to (that and the trackpad makes couch gaming great), but in the end, opinions are like assholes lol.

nefarius, 2018-01-04:
Since it's UMDF v2 I can still port it to kernel-mode and add HID features later on. For now I'll continue coding.

nefarius, 2018-01-04:
@mirh pardon my ignorance but are you on my Discord?

mirh, 2018-01-04:
> sixaxis

I could tell you there are developers waiting to hook some goodman driver for that. 
Pressure sensitivity - sure, I can conceed you. But thankfully there are like tens if not hundreds of ps2 games responsive to that. 

p.s. if official sony controllers sucks somehow someway.. It's not like you cannot pickup alternatives, still nonetheless "based" on the same *platform* (indeed, for this reason I have always felt retarded those comparing design when console warmongering)

@nefarius nope, I'm more of an irc/telegram whatever-that-has-an-open-api guy
Yeah, I'm pretty loquacious. 

p.p.s. maybe I should rename myself to KMDF-guy, given almost every feature I end up requesting leads there /s

nefarius, 2018-01-04:
Everything will be fine 😃

nefarius, 2018-01-04:
@JohnEdwa yes

fr500, 2018-01-09:
> FS v2 is an USB driver only, it won't expose HID-Capabilities and won't be visible by games.

Does this mean no pressure sensitivity nor rumble?
Just wondering, looking into hooking them up in RetroArch.