nefarius/FireShock #27: PC Slows down horrendously over time while using a PS Navigation Controller

tunbridgep, 2018-02-08:
Hi! I am using the PS3 Navigation Controller that came with the PS Move.

This driver is very good for getting this device working, and it works flawlessly in-game, except for one very major issue.

Whenever I try to play a game, and I use the controller in conjunction with the mouse (it is a 1 handed controller), my PC starts to slow down exponentially, and it eventually gets so slow and so laggy that the sound starts popping instead of sounding normal. Hard to describe, I know, and I can't record the effect unfortunately.

I am running Windows 7, using the latest version of the driver. My PS Navi is connected via USB

JoshWobbles, 2018-02-08:
Related to https://github.com/nefarius/FireShock/issues/4

nefarius, 2018-03-20:
I should really delete the releases. Reminder set.

mirh, 2018-03-21:
Those just work for DS3 tbh..

JoshWobbles, 2018-03-21:
Was the move or nav controller issue ever figured out? ive been AWOL for a while and missed a few things.

mirh, 2018-03-21:
Nope, total rewrite of the driver.. And crossed fingers.

nefarius, 2018-03-24:
The only media where up-to-date news is circulating is my Discord.

CareFudundae, 2018-04-17:
I had the same issue, but changing to a usb 3.0 port fixed it for me... not sure why. got this idea from: https://github.com/ViGEm/FireShock/issues/4#issuecomment-317292914

nefarius, 2018-04-20:
Since this issue originated from major version 1 release - which I have now officially deprecated - it's no more relevant and closed.

closed 2018-04-20 by nefarius