nefarius/FireShock #31: Support for DS2?

PrinceOfJudah, 2018-04-19:
Does this support DS2? I know scptoolkit did but it hasn't been updated in a long time. I have a ps2 controller but not a ps3 controller available to use on PC.

mirh, 2018-04-19:
Rather than DS2 specifically, I think you should specify which particular kind of adapter you are using.

JoshWobbles, 2018-04-19:
If you are using a ps2 to USB adapter you shouldn’t need this at all as the adapters present themselves to the pc as a direct x controller already.

JohnEdwa, 2018-04-19:
The two adapters I have present themselves as a "PLAYSTATION 3(R) controller"s just as plugging in a genuine DS3 does and as such, work with Fireshock.

One of them [looks like this](https://images3.pricecheck.co.za/images/objects/hash/product/24d/709/518/image_big_90721854.jpg) while the other [like this.](https://c.76.my/Malaysia/playstation-ps2-to-ps3-pc-controller-adapter-usb-converter-dance-doinfinity-1307-07-doinfinity@3.jpg)

PrinceOfJudah, 2018-04-19:
Uhm, I bought one off of ebay. it doesnt look like either. I also am broke completely now.

PrinceOfJudah, 2018-04-19:
https://www.ebay.com/itm/302279785360

JohnEdwa, 2018-04-19:
It should work the same way, it says in the description: "Convert your PS2 Dual Shock Controller into a PS3 Dual Shock Controller" - it'll make the controller look like a DualShock 3 so it would work on a PS3 or PS4, and it does so on a PC as well.

It's just any adapter that was made/designed before the PlayStation 3 was released (like [this one](https://www.ebay.com/itm/PC-Converter-Adapter-Cable-PS2-Handle-Converter-PSX-PS1-PS2-To-USB-Converter/162871278435?_trkparms=aid%3D888007%26algo%3DDISC.MBE%26ao%3D1%26asc%3D20131227121020%26meid%3D8868f0d4090e4aec8c53887fc4ec8573%26pid%3D100009%26rk%3D1%26rkt%3D2%26sd%3D302279785360%26itm%3D162871278435&_trksid=p2047675.c100009.m1982) for example) show up as a generic gamepad on a PC.

mirh, 2018-04-19:
The real® question, is how (if at all) analog buttons are exposed IMO.

PrinceOfJudah, 2018-04-19:
DOH. I cant believe i didn't see that.

nefarius, 2018-04-20:
No it doesn't.

closed 2018-04-20 by nefarius